manuelbl/usb-pd-arduino #6: STM32G474xx support

VIPQualityPost, 2024-01-29:
Hi,
I was trying to port the ST USBPD library to work with Arduino but discovered your project.
However, it seems you only support STM32G431xx. This part is not so different from the G474. I am working on developing some USBPD applications so I am using the G474RE nucleo along with the SNK1M1 shield, which should be compatible with your library (it just connects the CC lines to a dedicated USB port, along with a MOSFET for shutting off Vbus downstream of the port).
When building a bare UCPD project from CubeMX the connection to source works fine. With this library, I don't seem to get any response from the sink (with your library) when the source advertises PDOs. Do you have any suggestions to debug this? 
I have added define flags at all locations where there are guards for the G431xx to also include G474xx.

manuelbl, 2024-01-29:
What should the role of your Nucleo board be? Should it monitor the USB PD communication, be a power sink or a power source? What and how do you connect to the Nucleo board?

For debugging I've used mainly two techniques:

- Use some GPIOs and pull them high or low at interesting places in the code. Then connect an oscilloscope or logic analyzer to it and monitor it.
- Implement a non-blocking serial ouput and write debugging messages

VIPQualityPost, 2024-01-29:
The Nucleo is a sink for me, I just connect it to a PD source. I do have another device inline which is sniffing the packets but I get the same behavior in both the case of it connected and disconnected.
I'm not observing any goodCRC response so I was wondering if there is anything obvious about hardware configuration, but I suppose that you haven't looked closely at 474 :) , so I will compare the UCPD init with what I have working already to see if there is something fundamentally different (even as they are both G4 family).

manuelbl, 2024-01-29:
I made a quick comparison of the STM32 G431 and G474, and I couldn't find any obvious differences. The pins and the DMA channels seem to be the same on both chips. Timer 7 is available on both boards. And there isn't much more to configure.

However, when using STM32cubeMX I noticed that the pin PB4 is assigned to JTAG debugging by default. Could it be that the Arduino implementation for the NUCLEO-G474RE does the same? Could the Arduino framework be the problem?

VIPQualityPost, 2024-01-29:
It's possible, I just checked now also too.
I think for this board, the pinout of Arduino header is slightly different (D10 (PB6)and D5(PB4) rather than D12 and D6 for G431), but fixing this did not change anything. I wonder if it is the JTAG interface taking control.

manuelbl, 2024-01-29:
I've just discovered a difference. The CC1 and CC2 pins are first initialized via the Arduino framework, and then via the STM32 Cube LL routines. While the STM32 pins designations are the same (PB4 and PB6), the Arduino designations differ:

| USB | STM32 | G431RE Arduino | G474RE Arduino |
| - | - | - | - |
| CC1 | PB6 | D6 | D10 |
| CC2 | PB4 | D12 | D5 |

You might want to change them if you haven't done so yet.

manuelbl, 2024-01-29:
The JTAG hint is actually wrong. I've opened the wrong board in STM32cubeMX. There doesn't seem to be a conflict with your board.

Does your shield need some special configuration, e.g. activating the dead battery signals or some configuration of the pull-down resistors?

VIPQualityPost, 2024-01-29:
[image]
I also noticed that from this diagram in the Nucleo datasheet, the pins should not be different.. maybe this is a mistake on the STM32duino side. Or, you are using the G431KB in the smaller form (not RB in large board). 

There are two controls on the shield, one for Vbus enable (active high), and another for DB enable (active low), but I just leave these both high in my CubeMX project and it works - I am also initializing these and setting to high before PowerSink.start() in the Arduino project. I have also tried jumping CC1 and CC2 to respective DBCC1 / DBCC2 on the G474RE directly, but this does not change anything.

manuelbl, 2024-01-29:
I'm indeed using the G431KB in the smaller Nucleo-32 form factor.

manuelbl, 2024-01-30:
Today, I've tested it again with my device and it didn't work. It turned out to be a problem with the decoding the type of SOP sequence. It very much looks as if the underlying STM32 cube LL libraries have changed, and the code assumed properties about the LL constants, which are no longer given.

Anyway, I've fixed it and released version 1.0.3. I'll probably take some time until it is available in the Arduino library manager.

VIPQualityPost, 2024-01-30:
Thanks! This works for me now. I opened a PR to support these boards and clarify the Arduino pin naming for CC pins.

PR #7 (2024-02-02): Add G4 family support
Add support for other G4 Nucleo boards (G431RB, G474RE, G491RE)
Add support for building "standard" G4 family (G474xx, G491xx)
Use ARDUINO_NUCLEO defines to clarify pins used on dev kits for CC pins.
Closes #6.
closed 2024-01-30 by VIPQualityPost