manuelbl/usb-pd-arduino #12: Issues with some PD source

VIPQualityPost, 2024-04-23:
Hi Manuel,

I discovered that I have some "picky" PD source which do not play well with the library (work well with other devices, like laptop). 
I tried to check with my analyzer but did not find any sort of expected misbehaving with the CC communication. Have you seen this behavior before? It seems like the device is able to negotiate a power profile, but within 10s of ms it will drop the connection and the microcontroller will restart. I don't think this is related to inrush capacitance tripping the source, as some sources are fine (or perhaps my other sources are just too lenient with their output power?)

VIPQualityPost, 2024-04-23:
I just spent some more time digging, and confirmed with the same binary, this charger does not work , just bootlooping with any requested source voltage 9/15/20V(Anker 65W): 
https://a.co/d/0JOoG96
But this charger has no problem requesting stable 9V (does not have any higher voltage PDO) (Anker 33W):
https://a.co/d/3TmiCP5

VIPQualityPost, 2024-04-23:
Yes, this is with G431CBU6, using UCPD (no additional TCPP chips, just CC wire direct to cable + ESD protection)

manuelbl, 2024-04-23:
Do you have a way of recording the USB communication (incl. precise time stamps)? That would be very helpful.

An alternative is to add `PDProtocolAnalyzer.poll();` to `loop()`. It will also log the USB PD communication.

VIPQualityPost, 2024-04-24:
Yes, I can try to capture a trace, unfortunately I won't have access to more debug tools until next week. I will write back when I can save some data for you.