manuelbl/usb-pd-arduino #17: v2, FUSB302 USB PD controller and PPS

gc-ss, 2024-08-06:
Noticed you have a v2 in the works and have chosen the FUSB302 USB PD controller. It's my understanding that the FUSB302 doesn't support PPS. There are others, like the AP33772 that does support PPS, though.

Would you be interested in the AP33772 so that PPS can be supported as well?

manuelbl, 2024-08-06:
"v2" is an experimental branch. It's unclear if it will ever reach a reasonable state. The FUSB302 implementation works with ESP32 only. In its current form, it's not suitable for other MCUs.

FUSB302 fully supports PPS.

The AP33772 is – as far as I can tell – a far more autonomous PD sink controller than FUSB302. It fully covers all protocol layers while the FUSB302 only covers the lower layers. In order to use it, there is no need for handling all protocol details and timing constraints. It's sufficient to set a few I2C registers. Arduino examples are provided by the manufacturer. It doesn't really make sense to add support for it in this project.

gc-ss, 2024-08-06:
Thank You for the explanation and sharing your findings

> It doesn't really make sense to add support for it in this project

Can you explain this a bit more? It seems the AP33772 is more "user friendly" and you don't lose any functionality over FUSB302, so if you will be using an external PHY, why won't you use one that's cheaper and more "user friendly"?

manuelbl, 2024-08-06:
USB PD communication consists of several protocol layers. In simple terms, they are:

- Lowest, physical level: generating PD waveform, receiving PD waveform, detecting connected source/sink
- Middle layer: encode and decode data into bit pattern and vice versa, reacting to connection and disconnection events, acknowledging messages, detecting out of order messages etc.
- Highest layer: React to source messages, request voltage and current, re-request PPS voltages etc.

This library covers all three layers. If FUSB302 is used, only the highest layer and most of the middle layer is needed. If AP33772 is used, none of the layers is needed as the chip handles all three layers itself. And requesting a certain voltage can be achieved in several ways, most of which do not involve any USB PD elements.

So this library cannot really provide any added value for the AP33772. It can easily be used with any I2C library.