manuelbl/usb-pd-arduino #18: ListCapabilities example when already connected

benjinne, 2024-09-16:
Hi,

Is it possible to list capabilities when PD sink is already connected?
When I reset my STM32G0 with it connected, no capabilities are returned from the event handler since no new event was created.
If possible it would be a nice example to add.

manuelbl, 2024-09-17:
The library assumes that your device is a so called *bus-powered USB device*, i.e. it only gets power via USB. If so, your described behavior will not occur because the device will only (res)start when it is connected to a USB cable. And in that case, the USB-PD negotiations will occur and all events will be sent.

What is your planned used case? Will you have a device with separate power (so called *self-powered USB device*)? Or will your device restart as part of its normal operation? Or is it just during development?

Depending on the use case, one of several options must be implemented (resetting the USB-PD connection, likely leading to a temporary power loss; soft reset to synchronize the communication etc.).

benjinne, 2024-09-18:
My use case is a self-powered device.
I actually got it working using:
`PowerController.sendDataMessage(PDMessageType::controlGetSourceCap, 0, nullptr);`
In my setup() function, then resetting the MCU with everything powered and connected.
I'm using a nucleo g0 board with the X-NUCLEO-SNK1MK1
The stm32 gets power from the  micro USB port and the USB PD C stays connected and powered.
When I reset the STM32, it's able to read the PD capabilities